nion-software/nionswift #1361: Lattice tool Intermediary ellipse movement not funtional

lisham2000, 2025-03-11:
When creating a lattice only the two end ellipses are movable but the one connecting them (one with singular dot) is not movable. This means if that ellipse is not correctly positioned it cannot be moved to the desired location.

[image]

cmeyer, 2025-03-11:
The lattice is always centered on the calibrated origin. Is there a need for anything else?

lisham2000, 2025-03-11:
If this is the intended functionality and you would never want an off-centre lattice that is all good.  Assumed with the green centre marker it would be adjustable but if not this can be closed.

cmeyer, 2025-03-12:
As with the other Fourier mask tools, the lattice is symmetrical in Fourier space around the _calibrated origin_. So you can make it off-center by changing the calibrated origin. See also:
- https://github.com/nion-software/nionswift/issues/362