nion-software/nionswift #1372: "Go to Source" in a Combination of 2 line plots (Multiple data items) doesn't appear as an option on the submenu

Ion-e, 2025-03-14:
As title, with the setting shown in the following screenshot, please observe that right clicking in the Source thumbnail in the display panel doesn't bring up the option to "Go to Source "Generated Data".

[image]

cmeyer, 2025-03-14:
In this case, I'm wondering if we might name it 'Component' or something other than 'Source'. The overall naming might need work, but I've been thinking of having 'Source' and 'Dependent' refer to computation inputs/outputs. 'Component' when the display has several components. And 'Related Item' when there are related items, i.e. HAADF and SI would be related via the upcoming acquisition object. Thoughts?

cmeyer, 2025-03-14:
See
- https://github.com/nion-software/nionswift/issues/1020

Ion-e, 2025-03-14:
(Please mind that post-editing the title may affect the description of the ticket.)

This ticket is not related to the Naming convention for the Source/Dependent.

It states that  in composite line plots, there is an inconsistent(compared with 2D raster display) lack of the functionality  for the "Go to Source/Dependant.

I would exclusively encapsulate that behaviour in this ticket, furthermore, I would do a follow up on that, exclusively in this ticket.

cmeyer, 2025-03-14:
Yes I understand. I suggesting that for composite line plots, the context menu item names for referring to the component items of the line plot should be 'Reveal _Component_ "abc" in Data Panel' not 'Reveal _Source_ "abc" in Data Panel'

cmeyer, 2025-03-14:
Linked PR #1374

Ion-e, 2025-03-17:
Issue verified as failed, the "Reveal source/component" button is missing.

[image]

Rather than going for what would be technically correct naming, in this case I would prioritise keeping the functionality's name consistent with other items for easiness of use by the user.

Ion-e, 2025-03-18:
This is still an issue.

cmeyer, 2025-03-25:
@Ion-e This works for me. Can you retest with latest version?

Ion-e, 2025-04-03:
Verified as fixed. Thanks!

closed 2025-04-03 by Ion-e