nion-software/nionswift #1380: Colour palette issues

Ion-e, 2025-03-18:
If you go to the colour palette for the Fill colour of a transparent/semi-transparent graphic (Or any transparent or semitransparent object attached to the colour palette)

1. Access the colour palette. then press OK wihtout making any changes.
- Actual: the transparency (alpha channel) goes to 255 (full opaque).

Also, please notice that changing the Alpha channel doesn't change the HTML values. (Which could be considered a different bug).

[image]

cmeyer, 2025-03-21:
Let's resolve this by having the inspector open the dialog with the `used_color` and then only update it if changed. The longer term design might be to have our own color "chooser" dialog that allows the user to have more control over whether the color is specified or not and whether to use a standard palette or not; as a last case they might want to choose a specific color with the standard color picker dialog.

Ion-e, 2025-03-21:
To test once Graphics PR has been merged.

Ion-e, 2025-03-28:
I would disregard my latest comment. 
The graphics PR shouldn't encompass the behaviour of this window(nor should the current changes to the composite line plot default colour selection), should we make a new issue for the colour palette?