nion-software/nionswift #1391: Composite line plots to have "no fill" as default fill for all lines

jamesrussell216, 2025-03-24:
From Ben...

What I asked for is for the default display of composite line plots to be changed. As Tracy mentioned, currently when you create a composite plot one data item has a solid fill, and the other has no fill and it [the composite plot] is usually visually indecipherable. If there are two or more line plots, they all need to be unfilled lines. The default line weight should also be increased.

--

So coming out of this I think we want is...

- Default line profile plots (when there is only one data series) to remain the same as they are now; filled with dodgerblue, no stroke colour.
- Composite line plots to have defaults (for all layers) of... 
   - Fill Colour; None, 
   - Stroke Colour; [As now], 
   - Stroke Width; 2

KRLango, 2025-03-24:
Documenting the current plot behaviour that we are wanting to match:
When adding a new plot, if the fill color is default, then make the stroke color the next default in the list, if it is not the default color, make the stroke color match the fill color.

Question:
If you have existing lines that you have changed the fill/stroke on, when you add another line plot, do you want to keep those changes, or reset to fill color: None, stroke color: default, stroke width: 2?

Ion-e, 2025-03-27:
Current behaviour:
a. When creating a composite line plot from default line plots:
1. Dragged line plot gets added with standard Red stroke colour, None colour fill with full opaque(bug).
2. Base line plot remains with Dodgerblue stroke, with full opaque fill.
b. When creating a composite line plot from 1 default, 1 full customised plots:
1. Dragged customised line plot gets added with the customised stroke colour, None colour fill with full opaque(same bug as a1).
2. Base default line plot same as a2.
c. When creating composite line plot from 2 full customised plots:
1. Dragged line plot keeps stroke colour, but loses the Alpha channel customised value(bug). Fill same bug as a1.
2. Base line plot keeps customisation.

cmeyer, 2025-04-07:
Linked PR (merged) #1415

Close?

Ion-e, 2025-04-11:
Looks good to me!

closed 2025-04-11 by Ion-e