nion-software/nionswift #1397: set_metadata_value failure

luc-j-bourhis, 2025-03-27:
```
import numpy as np
from nion.data.DataAndMetadata import new_data_and_metadata
from nion.swift.model.Metadata import set_metadata_value

data = np.random.uniform(0, 1, (2,3,4,5))
dnm = new_data_and_metadata(data)
set_metadata_value(dnm.metadata, 'stem.high_tension_v', 200000.0)
```
crashes:
```
File "C:\_DEV\Nion\developer\main\nion-software-research\DIALS_IO\sandbox.py", line 7, in <module>
  set_metadata_value(dnm.metadata, 'stem.high_tension_v', 200000.0)
File "C:\_DEV\Nion\developer\main\nionswift\nion\swift\model\Metadata.py", line 187, in set_metadata_value
  metadata_source.metadata = d0

builtins.AttributeError: 'dict' object has no attribute 'metadata' and no __dict__ for setting new attributes
```

This kind of code may be somewhat atypical but `set_metadata_value` seems very useful to me to write test cases. An alternative is to create a dictionary `metadata` by hand but then one has to use the "low level" metadata keys after reading "key_map" in Metadata.py. And that defeats the purpose of the "high level" metadata keys in such tests.

cmeyer, 2025-03-27:
Link to PR:
- https://github.com/nion-software/nionswift/pull/1402

cmeyer, 2025-03-31:
This should be fixed now. Try again and close if it works?

Note: your last line in your example will not change metadata of the data item itself, but only change the values in a _copy_ of the metadata. If you want to change the metadata of the data item, you'll need to use the data item methods, e.g. `data_item.set_metadata_value(...)`

luc-j-bourhis, 2025-04-01:
That code works
```
        self.metadata = m = {}
        set_metadata_value(m, 'stem.high_tension_v', 200000)
        set_metadata_value(m, 'stem.scan.frame_time_s', 1.0)
        self.data_and_metadata = new_data_and_metadata(self.data,
                                                       metadata=self.metadata)
```
and that's all that is needed for testing.

Note that at Metadata.py:188, `has_session_metadata` is set but never used. Same at line 250 as far as I can see.