nion-software/nionswift #1411: Application will not exit with notification panel open on Windows

cmeyer, 2025-04-01:
Notifications from acquisition panel failure.

cmeyer, 2025-04-04:
@Ion-e This changes how the application exits when the last window closes. The change that previously, an open dialog (e.g. notification panel) would prevent application close when closed with the close box; now it doesn't. However, there are a lot of cases with switching projects, using the choose project dialog, or launching for the first time and needing to choose a new project. Can you go over those tests again? I checked over everything I could find and resolved all issues that I found. But another set of eyes would be useful.

closed 2025-04-04 by cmeyer

Ion-e, 2025-04-10:
This has an odd behaviour.

After opening the notifications and the choose project windows, and closing swift.

Swift closed (As seen it's not running on windows taskbar, however these windows remained open) 

[image]

cmeyer, 2025-04-10:
Yes, open new issue, maybe one for each type of dialog? Thanks!