nion-software/nionswift #1416: Deleting a live item in the display panel produces odd behaviour

Ion-e, 2025-04-03:
When deleting an live data item that's playing, the item gets replaced by an odd version of it, please see screenshots:

Before:
[image]

Deleted:
[image]

Please also note that in some occassions, the "odd live item" that replaces the original get's located in a display panel that's not the first one. (although it's been tricky to reproduce)

Ion-e, 2025-04-15:
Apologies about the confusion. The "odd" live item refers to this item in the following screenshot. 

[image]

Please note that this "replacement" item has:
1. No green banners.
2. No controls.
Please also note that: 
1. The item should be deleted correctly.
2. An edge behaviour appeared, where the "replacement" live data item is placed in a data panel that's not the first one available.

Please let me know if different tickets are needed for these behaviours.

jamesrussell216, 2025-05-22:
@cmeyer question; should we even allow users to delete live data items? What's the use case here?