nion-software/nionswift #1420: Next - Previous Workspace works inconsistently

Ion-e, 2025-04-04:
This tool seems to work inconsistently. To test and figure out STR:

Ion-e, 2025-04-07:
I haven't figured out any pattern I'm afraid.

It seems that, if no changes are made to the workspaces:
- It seems to keep a consistent order, which in my case it matched the project items order shown here:
[image]

- The order doesn't seem to follow any pattern besides the one shown in the project items.
- 
If changes are made to the workspaces (i.e: new workspaces are created):
- The order will change, and it won't follow anymore the project items order (Even after restart)

KRLango, 2025-04-07:
I had a bit of a play with this and found the current behaviour:

Next/Previous (effectively) work on time created. Which means that they will always go through the workspaces in the same order.

The list of workspaces is order by last modified, with the most recently modified at the top.

This means that the next/previous buttons can jump around the list depending on the order in which you modify the workspaces.

bentps, 2025-05-28:
> Yes I see the conflict here. I'd say we keep that list being ordered by
> last modified - that's how these sorts of lists commonly work. So I would
> say we move the next/previous functionality to just cycle through
> workspaces based on the same list - i.e. going to the next workspace should
> go to the +1 most recently modified, and previous should go to the -1 most
> recently modified.
>
> @bentps <https://github.com/bentps> @cmeyer <https://github.com/cmeyer>
> would be good to get your input on this one as you have more experience
> with these use cases?

A little while ago (more than a year ago) I asked Chris to keep the list of
workspaces sorted by creation time rather than last modified time. That
sort of worked for a little bit but was buggy, and then I think more
recently has regressed to last modified.

For my use of workspaces, I strongly desire them to be sorted by creation
time. I often switch back and forth between workspaces, and being able to
hold in my head an ordered list is possible, while having a constantly
resorted list is a nightmare. I'm not aware of anybody who uses a lot of
workspaces and who strongly desires the modified time sorting, but if you
can produce such a person I would consider their arguments. If not, please
sort by creation time. It sounds like that behavior needs a regression test
as well.

Sorry to mix issues, but I feel the same way about the data panel: there
should at least be an option to sort the data panel by creation time of the
data. It is again a total nightmare having that list constantly shuffling
and resorting.

jamesrussell216, 2025-05-29:
Thanks for input @bentps . I hear you on the data panel - we know we've got more work to do on that.

Ordering them by time creation is the conceptually simpler way so that works fine for me too. @cmeyer you OK with that?

closed 2025-10-15 by Ion-e