nion-software/nionswift #1428: Changing annular ring redoes all the dependent computations

Ion-e, 2025-04-07:
When having a display item with:

a. Annular ring1 with no dependent computations.
b. Line profile with dependent "A".
c. FFT as Dependent "B":
        - Annular ring2 on dependent "B" and a mask from it generating dependent "C"

If the user changes annular ring1, dependent "A", Dependent "B" and dependent "C" will redo their computations. This will also cause them to update in the data panel. However if Line profile is changed, only dependent "A" will change.

cmeyer, 2025-04-07:
Steps to reproduce:
1. Generate 1024x1024
2. Perform FFT
3. Add annular ring filter to the FFT
4. Add line profile to the FFT
5. Open Activity panel to observe computations
6. Drag annular ring filter
7. Observe computation activity for line profile

cmeyer, 2025-04-08:
Link to PR
- #1428

cmeyer, 2025-04-09:
I had incorrectly made the PR comment above link back to this issue itself; fixed that. PR is #1430. It has not been merged as of this comment, but will do so shortly.

closed 2025-04-09 by cmeyer

Ion-e, 2025-04-11:
Fixed now, thank you very much!

commit fba564b1 (2025-04-09): Fix #1428. Reduce unnecessary computation recomputes on unrelated changes.