nion-software/nionswift #1429: Mask from annular ring failed on 2D raster display

Ion-e, 2025-04-07:
Generate a data item(Not an FFT) > Create an annular ring > Generate a Mask from it. < Observe the mask fails. 

Changing the size of the annular ring, or creating a new one doesn't seem to solve the issue. I will keep the data item in case we will need it

[image]

cmeyer, 2025-04-07:
Can you provide more details of the original image size and the specific ring parameters that cause this to fail? I cannot get mine to fail.

cmeyer, 2025-04-08:
Steps to reproduce:

1. Generate 1024x1024
2. Add spot mask
3. Move spot mask so that center is on top of right side of data item
[image]

4. Observe error