nion-software/nionswift #1435: "bounds_constrained" graphics can still get out of bounds when changed through inspector

Brow71189, 2025-04-10:
Our graphics have a property `bounds_constrained` which does not allow users to drag them outside the bounds of the containing image. This works fine for dragging the gaphic, but when changing its size and position by typing numbers into the inspector, you can still produce out-of-bounds graphics.

cmeyer, 2025-04-10:
This is intended behavior. They're also able to get out of bounds by scripts setting the values.

It's very difficult to decide how an inspector would behave with constraints. As a simple example, if the user sets an 'x' position to an invalid position, should it (a) ignore it, (b) change 'x' to nearest in bounds position and also change width (c) keep 'width' constant and change 'x' to be 'right - width'? This is one of dozens of similar questions.

Instead, I propose that clients should never assume that just because bounds of a graphic are constrained, that it will actually return a constrained bounds. I know this is counterintuitive, but it leaves the decision of how to handle it to the client.

Graphically it's much easier since the user by dragging a particular corner is indicating what part they area changing: either the position or the shape.

Brow71189, 2025-04-14:
I think a graphic that is "bounds_constrained" should not be able to go out-of-bounds by whatever method. Same for shape locked and position locked graphics btw.

To me it is obvious that it should jump to the nearest "allowed" location if you enter an out-of-bounds location into the inspector.

cmeyer, 2025-04-14:
I don't think you addressed the question as to what specifically happens when the user adjusts the inspector to an invalid value. What does "nears allowed position" mean?

Brow71189, 2025-04-15:
If I type "-5" for a position that starts at "0", I would expect it to jump to "0". I don't really understand what is so hard about this. Isn't this just normal behaviour for software that has has limits on some values? Maybe I'm missing something here, but to me there is no ambiguoity here and any different behaviour would feel strange as a user .