nion-software/nionswift #1439: Some of the editable units are not actually Editable

Ion-e, 2025-04-10:
From https://github.com/nion-software/nionswift/issues/1041

Current list of found cases:

- [ ] Edit > Preferences > Acquisition
- [ ] Edit > Preferences > Drift scan settings
- [ ] Acquisition Window > Multiple
- [ ] Acquistiion Window > 1D/2D ramp
- [ ] Inspector > Image Data > Brightness/Contrast/Gamma (No units shown here, but user can actually input letters into these edit boxes)
- [ ] Inspector > Calibrations > All the units here (This one is a bit tricky, although you can edit the units, would a user ever want to?, supposedly a user should have all the factual calibration, please correct me here)
- [ ] Inspector > Index > Any index (No units shown here, but user can actually input letters into these edit boxes)
- [ ] Inspector > Graphics > Graphics

jamesrussell216, 2025-04-17:
Just a note from me: The issue is if you try to manually type in different units to the (for example) ms (to say us) it just changes the units back to ms.

jamesrussell216, 2025-05-22:
Further note: Two ways I can see to solve this...

1) Change these controls to follow the fixed unit edit box standard format used elsewhere in the software, as per this example...

[image]

 2) Alternatively we can take the approach found in the inspector which basically just ignores the problem as per these example @Ion-e has raised...

[image]

I don't really mind either way. 1) is probably a better solution but this isn't exactly a major issue either way. @cmeyer any preference?

jamesrussell216, 2025-05-23:
@cmeyer nope no specific reason to fix now and if it's going to take a bunch of time to fix then maybe there's better bang for buck.
But I do want to start chipping away at these sorts of problems generally. As I said on the other thread, the problem is the quantity of them adds up to the point it effects the software usability. And the only way we progress it is by chipping away at them.