nion-software/nionswift #1440: Custom script expressions should be stored in target computation description (metadata)

cmeyer, 2025-04-10:


Ion-e, 2025-04-11:
From https://github.com/nion-software/nionswift/issues/1436#issuecomment-2794418128

Ion-e, 2025-05-22:
From https://github.com/nion-software/nionswift/issues/1436#issuecomment-2794418128:
Steps To Reproduce: 
1. Do an acquisition
2. Run a processing on it
3. Observe the name of the computation on the metadata
4. Control + Shift + E on the data item and change the name of the computation
5. Observe that the name of the computation in the metadata doesn't change.

Happy to demonstrate it's STR. However please note that not only the name of the custom computation script should be tested but also the computation parameters.

jamesrussell216, 2025-05-28:
OK so are we looking to have the above change the "computation_title" metadata field?