nion-software/nionswift #1441: New Project/Choose project windows don't close when closing swift

Ion-e, 2025-04-11:
From https://github.com/nion-software/nionswift/issues/1411

I can see the use for this, although I'm unsure if this will make the software to get into an odd state.

cmeyer, 2025-07-24:
If the user opens the new or choose project dialogs and then closes the main window, should those dialogs close?

Unless there is something preventing the application from exiting after all open windows are closed, I don't think this is an issue.

Ion-e, 2025-07-24:
That it's an interesting question. I think it should. As a user, if I close the "main" window I would expect the software to close down completely, so that, if I get into a weird state, I can make sure I have a fresh restart.

And I think this raises a couple of questions, does swift ever completely finish running on the background? If yes, do I need to close all the opened windows for it to close?

For the first answer I would expect a yes for the aforementioned reason. And for the second, I would expect just closing the main window should do it, as I personally have encountered rogue windows in corners of the screen or behind other apps.

Thoughts?