nion-software/nionswift #1453: Composite line plots change constantly

Ion-e, 2025-04-16:
When moving the line profile, sometimes the "Y" axis changes to display a bigger range in the composite line plot, whilst this is fine, this also happens when the values changed are still within the range.

Please observe following screenshots:
1. [image]
2. [image]
3. [image]

cmeyer, 2025-04-16:
As usual, please provide steps to reproduce. But overall, this looks like it is working as expected. The y-scale if left unspecified will adjust to accommodate the data.

Ion-e, 2025-04-17:
Closing as per comment. Thanks for confirming!

closed 2025-04-17 by Ion-e