nion-software/nionswift #1463: Rename data group menu option missing

jamesrussell216, 2025-04-30:
Currently you can add new groups, but there's no menu option to rename them, so you just end up with a bunch of "Untitled Group" groups. The code seems to be there under the hood to do, it's just not accessible from the UI. I suspect that the functionality to delete a group may fall into the same catagory!

jamesrussell216, 2025-05-02:
Just realised @cmeyer I never sent this one your way when we were on the train!

jamesrussell216, 2025-05-14:
Probably needs a delete function as well.

cmeyer, 2025-05-15:
This should be available by
- click-hovering the data group, like Explorer rename
- menu item when data group is selected
- right clicking data group
- pressing return when data group is focused

Any of the actions above should pose a small pop-up dialog to rename, similar to title edit. Pop-up dialog should respond to escape and return keys to cancel and accept.

cmeyer, 2025-06-06:
Can this be closed in favor of the more specific
- #1486

closed 2025-06-10 by Ion-e