nion-software/nionswift #1466: Option for default display mode for 2D x 2D data to be mapped

cmeyer, 2025-05-06:
The current display mode for 2D x 2D data is to provide sliders for each 2D collection dimension and then display the 2D datum at the point specified by the sliders.

The proposed change would allow a reduction of the 2D datum to a scalar value and then the resulting displayed image formed by that scalar value at each 2D collection point.

In order to provide this option, the `DisplayValues` will have to be changed to handle a potentially significantly slower algorithm. Even if the algorithm is as simple as 'take the center pixel value' it will still have to access a lot of memory to form the image.

Possible algorithms for reduction include using the center point, summing/averaging the entire image, or sum-multiply within a mask.

Additionally, this may have implications for how scripting handles display-data. Needs design.

jamesrussell216, 2025-05-22:
Some words from Ben...

Answering James question about center pixel vs tcBF: I think the problem here is making the user wait for the data to appear while some computation is being done before the data even shows up. I'm pretty sure that displaying the center pixel value from each diffraction pattern will be faster than any reasonable tcBF processing we can do. The point is to show the user the minimal useful thing so they can see that the data is even a valid image, then the user can decide what level of processing to apply. The current system just shows one diffraction pattern, so the user has no idea what they have.

I'm a little surprised that this is more complicated to do than the SI case which is already implemented; computationally it seems like exactly the same thing. An SI is shown by default as a 2d image in scan coordinates, showing the first pixel from each spectrum. While we are on this topic, it would also be an improvement to change the default SI pixel from first pixel to the nominal ZLP position (100th pixel on the Dectris, for instance).

Why is the 4D case more complicated than the 3D (SI) case? Is it much more complicated to show the Nth pixel of the spectrum or diffraction pattern compared to the 1st (I guess we should call it the 0th)? It wouldn't be ideal, but taking the 0th pixel from each diffraction pattern would still be a big improvement over the current system, if it is much faster and easier to implement.

cmeyer, 2025-05-22:
You can "prototype it" with the code below, which triggers slow display value computation on any image with brightness < 0.

Suggested things to try:

Start two live images (scan, camera) running at full speed. Then change the brightness of one of them to be < 0 in the inspector. System responsiveness is really bad, i.e. unusable.

Stop the one with brightness < 0. As long as display isn't recalculated, system responsiveness is ok.

However, I don't think we can guarantee these things won't be recalculated. For instance, if we want to allow the user to change which pixel is used or otherwise change the display computation. Or if we want to display the result during acquisition (which is another project, but coming up). Either of these situations will lead to responsiveness as in the two live images.

This experiment also shows how slow display calculations may be affecting the overall responsiveness of the system even in our existing use cases, too. Resolving the general case might be a big improvement overall, even for the non-4D stem case.

```
diff --git a/nion/swift/model/DisplayItem.py b/nion/swift/model/DisplayItem.py
index f852995a..9266c89d 100755
--- a/nion/swift/model/DisplayItem.py
+++ b/nion/swift/model/DisplayItem.py
@@ -691,6 +691,9 @@ class TransformedDisplayRangeProcessor(ProcessorBase):
         adjusted_display_range = typing.cast(typing.Optional[typing.Tuple[float, float]], self._get_parameter("adjusted_display_range"))
         brightness = self._get_float("brightness")
         contrast = self._get_float("contrast")
+        if brightness < 0.0:
+            import time
+            time.sleep(1.0)
         assert adjusted_display_range is not None
         display_limit_low, display_limit_high = adjusted_display_range
         m = (contrast / (display_limit_high - display_limit_low)) if contrast > 0 and display_limit_high != display_limit_low else 1
```

bentps, 2025-05-28:
> Quick Note:
>
> So the problem comes during the acquisition of the data item the
> calculation would essentially get re triggered each time the data item
> changes which would slow down the whole acquisition to the point of
> failure. Particularly because there will be lots of read/write on the HDD
> during that process. We could look at separating it out so that the code
> doesn't execute during acquisition, but that essentially takes you back to
> where we are the moment where you're using the 4D pick tool.

I was considering looking at this the other way: during acquisition, we do
various live processing of the data. What if part of the acquisition was
extracting and storing the center pixel value of the detector, and adding
that to the live-updating image. You certainly don't want to be retrigging
extracting all previous values when a new frame comes in.

At the moment, live display of Dectris data for in-progress acquisition is
effectively not working, because only the first frame is displayed. It
would be nice to get that working, but that is a separate issue. The point
is that if fixing the post-acquisition display of 4D data creates some
problem for live display of incoming 4D data, you should note that the live
display is already not working, and making it work could be done in the
context of the new static display behavior.

In general, the principle that I am aiming for is to display something
useful as quickly as possible, and not recompute it unless the user asks
for it. The initial display can be a special case that is computed from the
original streaming data. Once the customer activates a pick region on the
data, that would bring up a second data item showing the diffraction
patterns and eventually recompute the scan space display (displacing the
initial view). Modifications of the pick region on the diffraction pattern
would eventually trigger a recomputation, but the UI of this should be
handled carefully given the possible long computation times.  Does that
make sense?

jamesrussell216, 2025-06-05:
I might be misunderstanding you @bentps or just extrapolating what you've said but...

Are you suggesting we essentially have a new data item that comes out of a 4D acquisition - a 2D item, similar to a HAADF where the pixel values of said item are the centre pixel value of the DP. That gives you your quick useful display in this new item, retains the 4D item for post-acquisition processing etc, and avoids a lot of the technical risk of doing it with the 4D item that we're worried about. So you'd have something like this (and please excuse the limitations of usim here)...

[image]

bentps, 2025-06-06:
Not quite, I'm not suggesting a separate data item. The 4D data should be
represented by a 2D view, and I'm suggesting that the initial 2D view
should be calculated on the fly during acquisition. Post acquisition, if
you trigger it, the 2d view is recalculated in a potentially slow way that
we want to avoid doing initially.

From a user experience perspective, what I described above is how EELS
works.