nion-software/nionswift #1469: Unfocused text in line plot intervals should not be dimmed

cmeyer, 2025-05-08:
There should be some indication that it is not focused; but the text should be visible as these are used for information display and dimming them makes them difficult to read.

cmeyer, 2025-06-24:
I don't think we should change the behavior when the interval is not selected. It is way too noisy if text is drawn for all intervals; and the user has an easy way currently to decide if the text is drawn or not: select the interval or don't.

As with the original issue: when the interval is selected but the focus is not on the display panel, we should simply not dim the text.

jamesrussell216, 2025-06-24:
> I don't think we should change the behavior when the interval is not selected. It is way too noisy if text is drawn for all intervals; and the user has an easy way currently to decide if the text is [...]

As I understood it @cmeyer that behaviour was specifically requested by the customer though? If we're going to change an agreed spec, and the works already been done, then I would at least like to go back to the customer to clarify.

KRLango, 2025-06-24:
@jamesrussell216 I think that comes from the concern I raised about how it looked in the PR: https://github.com/nion-software/nionswift/pull/1579

jamesrussell216, 2025-06-24:
Ahh OK that makes a bit more sense. Yes I see the problem - but is that a realistic use case though? I mean, I can shove a whole bunch of graphics done on a display item and make it a right mess too, but if it's not how people actually use the software in practice we maybe solving a problem that doesn't exist.... so I think I'll drop the customer an email anyways and just get their two pence on it and feed that back, thanks for raising both.