nion-software/nionswift #1486: "Edit group name" collections window can only be closed by pressing enter

Ion-e, 2025-05-27:
Inconsistent with other windows. 

[image]

Ion-e, 2025-06-10:
Specs: https://github.com/nion-software/nionswift/pull/1481

jamesrussell216, 2025-06-13:
For my part, I would like to see this move towards a standard File Explorer edit name functionality like this...

[image]

Ion-e, 2025-06-13:
That solution could cover elegantly the scope of this ticket.

cmeyer, 2025-06-19:
I changed the 'get string' dialog to accept any time user un-focuses the input field. This is available in nion-software/nionui@0c93a9c3ed9f9752a93016ba6005b1ab0b11b97f or later.

Ion-e, 2025-06-19:
Issue verified as fixed, Esc is the only keyboard that cancels any change. Looks great now.

Thanks!

closed 2025-06-19 by Ion-e