nion-software/nionswift #1500: Pressing "Intro" in a 1D line plot resizes it

Ion-e, 2025-06-02:
As title:

Generate a 1D line plot:

[image]
Press "intro":

[image]

Why is it resizing it? Sometimes it makes it they Y scale wider or narrower. 
The data remains the same, it's the axes metrics that change.

cmeyer, 2025-06-02:
What is the "intro" ?

cmeyer, 2025-06-02:
To reproduce:

1. generate 1d data
2. press <enter> key
3. observe display limits changing

cmeyer, 2025-06-02:
After investigating, see 
- https://github.com/nion-software/nionswift/pull/1506

When on 'auto' the display limits are chosen so that nice axes are displayed.

When the 'enter' key is used, the display limits are set to min/max +/- 20%.

Bottom line: this works as designed.

Ion-e, 2025-06-03:
Closing as per comment, thanks!

closed 2025-06-03 by Ion-e