nion-software/nionswift #1507: Data item recorder doesn't record Metadata & doesn't have create an appropiate data item's name

Ion-e, 2025-06-03:
Two issues:

1. No metadata is saved.
2. The data item's title created uses "Recording" even after it's finished the recording of it. (Although you could argue that "Recording" is being used as a noun, the wording can still be confusing.)

jamesrussell216, 2025-06-09:
@cmeyer @Brow71189 what's the purpose of the Data item recorder?

Brow71189, 2025-06-12:
It is meant for recording a changing data item. It is intended as a general tool, that can create a video of any changing data item, even if it is not directly recordable by the acquisition panel.

jamesrussell216, 2025-06-13:
Just so I understand (and apologies if this is a tangent) what's the use case for this?

Ion-e, 2025-06-13:
Does the item data recorder records the data/display item, or does it record the whole of the Swift UI?

cmeyer, 2025-06-13:
Brief, but: https://nionswift.readthedocs.io/en/latest/user_interface.html#recorder