nion-software/nionswift #1536: Add computation fails when attempted on data items of different sizes

Ion-e, 2025-06-05:
Unsure whether this is an actual issue, would we expect this computation to work? 

[image]

However there seems to be an odd extra behaviour:
1. From a 1D line plot you can add it to an interval graphic in itself. In this case the computation result seems to fail sometimes but sometimes it fails, it seems to depend on where you drag the interval graphic showing a result similar to the following screenshot:

[image]

cmeyer, 2025-06-06:
It could produce a better error message, but works as expected.

cmeyer, 2025-06-09:
What operation would you expect to occur when doing a pixel-by-pixel addition on data of different shapes?

Ion-e, 2025-06-09:
Yes, not an issue as per comment. Thank you.