nion-software/nionswift #1540: Deleting display also deletes resulting computation data item

Ion-e, 2025-06-06:
1. Create a data item.
2. Create an ellipse graphic in the data item.
3. Create a display copy of that data item.
4. Run a "Crop" computation.
5. Delete the first display item.
6. The computation will disappear.

Expected: the computations should remain, even if the graphic used disappeared. (Maybe the graphic should be referenced in the computation?

cmeyer, 2025-06-07:
I can't reproduce this. I also encountered the `Computation Error` and I'm very interested in having a repeatable error case so I can fix it.

Ion-e, 2025-06-10:
There are 2 issues discussed in the original ticket, and a new third issue. Which one is the one you struggle to reproduce?

1.
> "Expected: the computations should remain, even if the graphic used disappeared. (Maybe the graphic should be referenced in the computation?"

I believe this one is explained in detail in the aforementioned steps to reproduce, I have no problem reproducing it following the steps.

2. 

> The "NoneType" object has no attribute "cropped_display_xdata" error.

I haven't been able to reproduce this error.

3.

> A new inconsistency found in how graphics/computations operate.

Instead of following the aforementioned steps, do the following:
1. Create a data item.
2. Create an ellipse graphic in the data item.
3. Add the ellipse to the mask.
4. Create a display copy of that data item.
5. Mask the ellipse. (Processing > Arithmetic > Mask)
6. Delete the first display item.
7. The computation will **not disappear.**
8. The following error will appear (fairly so in my opinion, however this third issue addresses the inconsistent behaviour with computations.

[image]