nion-software/nionswift #1591: Dependant from multiple source has different behaviours when changes are applied to the source, depending on it's source's order

Ion-e, 2025-07-11:
From the steps of the ticket #1590 

Identical changes to identical source data items don't always have an identical output on the dependant:

1. This only happens if the steps from #1590 are done on the first selected data item when running the computation. If the step "change the "adjustment" from "none" to "log"" is done on the second item selected when running the computation (The +1 item) then the calibrations of the dependant won't change.

cmeyer, 2025-10-04:
This issue is unclear to me. Can you elaborate, close, update the title, and/or include exact steps to reproduce and expected behavior in the top comment for this issue?

Ion-e, 2025-10-06:
The issue is that "A computation from multiple sources" doesn't always behave consistently when a change is introduced to one of it's sources (Seemingly dependant on which source is chosen first when the computation was run).

This issue, as far as I know it's still an issue that hasn't been addressed. However the steps to reproduce given in https://github.com/nion-software/nionswift/issues/1590 are no longer valid as that issue has already been fixed. However it may be the case that the scenario described in this ticket can happen with other computations.

Steps to reproduce:
- Get a version from before https://github.com/nion-software/nionswift/issues/1590 was fixed.
- Create GeneratedItem1
- Create GeneratedItem2
- Click GeneratedItem1 and GeneratedItem2 in that particular order.
- Run Make RGB computation and rename to RGB1.
- Click GeneratedItem2 and GeneratedItem1 in that particular order.
- Run Make RGB computation and rename to RGB2.
- Change the "Adjustment" in the inspector in GeneratedItem1 to "Log".
- Observe RGB1 loses it's calibrations but RGB2 doesn't.
- Reset the Adjustment back to "none" for GeneratedItem1.
- Change the "Adjustment" in the inspector in GeneratedItem2 to "Log".
- Observe RGB2 loses it's calibrations but RGB1 doesn't.

Please note that these steps take in consideration that the other bug is already fixed and the issue is unrelated to that fix, but if it's otherwise let me know.

cmeyer, 2025-10-06:
Issue #1590 was that an internal part of the data flow was not propagating calibrations; now it is. So I don't think this type of bug can occur. There may be other situations where calibrations are not propagated, but I couldn't find them when I fixed #1590. I'm not sure what to do next with this issue. It seems fixed.