nion-software/nionswift #1616: Focus gets lost when using tab key

Ion-e, 2025-07-23:
From https://github.com/nion-software/nionswift/pull/1552#issuecomment-3052540687

There seems to be cases where "focus" on the display panel gets lost. Unsure what the cause root seems, however it seems to happen when alternating quickly between using tab and using the 2x2 grid divider tool from the toolbar. particularly if the chosen display panel is a 1d line plot. This has low reproduce ability rate, but there are scenarios where the focus jumps from display panels to outside. It also seems to be reproduceable consistently if you manually move a data item from one location to another by dragging the title bar of it's display panel and then you press Tab (in this last case it tends to happen often the focus is lost and it changes to the inspector or something else)

cmeyer, 2026-09-09:
This was probably fixed along with #1954 or #1955. Recheck?

Ion-e, 2026-09-21:
Issue verified as partially fixed. There are still some odd behaviours including but not exclusive such as:
1. "Tab" from an edit box in the Scan control panel. It will end up leaving that panel.
2. If the ouput window is open, it will end up in the text field and just keep inputting "Tab" in the text field.
3. Using "Tab" in the data panel don't iterate trough the items unlike in the display panel.