nion-software/nionswift #1617: Tab key groups need to be created or fixed

Ion-e, 2025-07-23:
Unsure if it's already implemented.

From https://github.com/nion-software/nionswift/pull/1552#issuecomment-3052540687

The data groups haven't been created fully. The "display panel" data group seems to have been created, however pressing "tab" will still iterate between different windows instead of iterating through the items of the selected window. This decision was made last time we had a conversation with Matt, if my memory serves correct.

cmeyer, 2025-10-04:
What expected behavior do you think is missing? Steps to reproduce?

Ion-e, 2025-10-06:
I'm quite certain both the title and description of this ticket were dictated by you on one of our meetings. 

The tab groups should remain within it's own group, so we don't end up in a situation where by pressing tab we navigate to different windows. We discussed working towards it, as we now have in the graphics and in the display panels. This ticket was a reminder to introduce this in all the other windows. If it needs more discussion we can bring it to a meeting. :)