nion-software/nionswift #1647: Add scale marker preference for default position/color/background/style

cmeyer, 2025-09-03:
The user can choose scale marker position, text color, background in the inspector. However it is tedious to do that for every display. If we have a preference for the default, they can leave the inspector settings alone (unless they want to override) and change the default for all displays. This is not perfect since we may want to allow them to do this on a individual user basis, or a project basis, or choose from a set of styles, but this at least gets them initial functionality. The preferences are a computer specific preference. I'm also open to other suggestions about how to handle the preferences.

Leaving unlabeled until this is reviewed by the team.

ndellby, 2025-09-03:
We don't actually have user profiles much, we tend to be a few people working together on a project with the data going in a library, The natural thing would be for preferences to go with the library. That way if someone moved the library from a microscope to an analysis computer they would come along.

Ion-e, 2025-12-03:
Has this been completed already? @cmeyer