nion-software/nionswift #1656: The selected project doesn't open

Ion-e, 2025-09-19:
Copied a copy of a previously used project.

Attempted to opened such project.

Failed to open project.

cmeyer, 2025-09-19:
How does it fail to open? With an error message dialog? Or something else?

Ion-e, 2025-09-25:
It shows an error message. Also note that the copied project was in the same folder. 
After some testing, I'm almost certain that the issue I was seeing is a behaviour caused by using UUID names, for which I believe we already have an issue and unsure on whether it's an actual bug.

Closing this as dupe, shout if you have any thoughts. 
[image]

closed 2025-09-25 by Ion-e